tum-autonomousdriving/autonomous-driving-simulator #1: Of one timestep all 3 photos are the same

Sodamin, 2023-05-29:
When running simulator_train/simulator.exe the 3 exported images in Data are the same for a single timestep.

[image]
[image]
[image]

--------------------
System:
Win 11

Einholz-Gao, 2023-05-30:
Oh， Thanks！In next Version I will fix it.

Einholz-Gao, 2023-05-31:
[image]

To address the synchronization issue, I merged three images into one output, and the performance of the simulator has also improved. I have a question, though. It shouldn't be difficult for you to handle the data preprocessing, specifically splitting this combined image, right?

Sodamin, 2023-05-31:
Nice one - super fast! Year splitting it should be fine :) 
Thanks!

closed 2023-05-31 by Sodamin