tum-autonomousdriving/autonomous-driving-simulator #2: [Major Bug] Only 8 images get created per second

Sodamin, 2023-05-29:
I ran the simulator once for 30s and once for 60s it created just 8 Image sets per second. I think this is probably quite few compared to the Udacity simulator and might lead to bad control performane?

------------------
System
Win11
AMD Ryzen 7 7700X 8-Core at 4,5 GHz
RTX 3090

Sodamin, 2023-05-30:
So the current ones from udacity are 320 × 160.
So fa the resolution was enough, but we might get problems with traffic lights.

Maybe we can for now keep the resolution and we should downsample all images if we need less. only if you need to go down because of the performance - and then I think it won't be a big issue

Einholz-Gao, 2023-05-31:
I have already increased the sampling frequency, but unfortunately, I cannot go beyond 20 frames per second. This is because I need to strike a balance between the performance of the simulator and the sampling frequency.

Sodamin, 2023-05-31:
Hmm I understand. Just a small one: I noticed motion blur, maybe removing that is just another small step to better performance?

Also: I am interested how others are doing with weaker hardware if someone can comment here

Einholz-Gao, 2023-06-01:
Yeah now I close the Motion blur. Thanks for the suggestion!
And: I build this project with a small window size(640*480), and hope it can run smoother on weaker hardware. If it's too small for you, please write to me.

closed 2023-07-19 by Sodamin