tum-autonomousdriving/autonomous-driving-simulator #5: Steering Controls

pranav-ap, 2023-05-30:
Hey,

Are the steering controls working as intended? Maybe its just me, but driving around and collecting data is very difficult for now.

1. Sometimes it is very sensitive to our inputs and makes large turns. 
2. It is difficult to stop the vehicle in traffic. It almost always reverses and hits the car behind.

Can you also support using arrow keys for driving.

Thanks

OS - Windows

Einholz-Gao, 2023-05-30:
1. because with keyboard the steering value is only 1 and 0. So next step I will try to fix it. But considering that the control logic comes from the Physic engine. it may cost some time. As an alternative you can try **Joystick**(if you have one), it will be better than keyboard.
2. right... because the key "s" is not only a brake, but a reversing too. I will change it into another control logic. 
Thanks for report this!

Sodamin, 2023-05-31:
If it is easier, using the mouse drag in window to change the steering really worked very well for Ferdinand and me.

pranav-ap, 2023-06-01:
Oh. Am we supposed to hold another key before doing mouse drag? what is it?

Einholz-Gao, 2023-06-01:
> If it is easier, using the mouse drag in window to change the steering really worked very well for Ferdinand and me.
Yeah. Next version I can implement a control method with mouse. 
At the end of this week the update will come!

pranav-ap, 2023-06-06:
I just tried the windows build. The screen is just black. The scene is not loading. Let's see if others also face this same problem.

closed 2023-06-28 by pranav-ap