tum-autonomousdriving/autonomous-driving-simulator #8: Windows Simulator Doesn't Load - Black Screen

BaranEkin, 2023-06-08:
On Windows 11 system, when Simulator.exe is clicked, small window appears with "made with unity" intro and then it shows just a black window and nothing happens.

Sodamin, 2023-06-08:
+1 
Same for me, just having this greyish background after logo.

pranav-ap, 2023-06-11:
+1 
This problem still persists in my Windows system with the latest sim version.

Einholz-Gao, 2023-06-12:
Thanks for reply, I will try to fix it again. V0.1 can be opened.

Sodamin, 2023-06-12:
correct, some more debugging help: the recording starts but even the recorded pictures are black. i can also share my screen with you if you want

Einholz-Gao, 2023-06-15:
Now it works well, right?

ge49nuk, 2023-06-18:
For me it works now!

pranav-ap, 2023-06-18:
Thanks for the fix

BaranEkin, 2023-06-18:
Yes, it works, thanks!

closed 2023-06-18 by BaranEkin