tum-autonomousdriving/autonomous-driving-simulator #11: [Feature Request] Return current speed in test mode

Sodamin, 2023-06-14:
Seems like we get steering angle and throttle back from the simulation.

I would propose to rather send back at least the current speed. To allow to control for specific speeds. (We should probably also think about removing the speed limit within the simulation in test mode)

Einholz-Gao, 2023-06-15:
Yeah, as you can see I add a speed limit module because the speed of this race car are too high without a limit. I don't think we should remove it in test mode, before we change a normal, slower car...
You can try removing the speed limit and have a look, it will run fxxx so fast...

tarikisildar, 2023-06-23:
I think he is asking about returning the current speed information as a data like steering angle and throttle. I agree it's a must have.

Einholz-Gao, 2023-06-24:
Hi.
Right, so do you want to send the speed back to python or just display it on the screen?
Now the speed control depends only on the throttle and brake. In Unity we have a speed limiter, too.

Sodamin, 2023-06-25:
I think what would make sense is that the simulator gives it back, just like the image. 
Because what a car should do is also to regulate the speed by using the throttle to accellerate until the desired speed (which might be the speed limit, but maybe also below when going around corners etc.) So it is essential to have a closed speed regulation loop.

Einholz-Gao, 2023-07-12:
Next version we can get the velocity from SocketIO in python

BaranEkin, 2023-07-19:
Are we getting this feature soon? I also agree it is very fundamental and absolutely required.

Einholz-Gao, 2023-07-20:
After Testing this velocity value can be sent through python code. Please after the version 0.24 updating check the py code and try it on the new simulator.

tarikisildar, 2023-07-20:
Just to clear any possible missunderstanding, the request is the simulator sending the velocity information out (to the python code) during the runtime. simulator itself doesn't require any speed input. Training data also doesn't (this maybe can be used but probably not).

Also the need for the speed-limit within the simulator comes from the car accelerating too fast in my opinion. If we can tweak the engine torque and gear settings it'll become more managable.