tum-autonomousdriving/autonomous-driving-simulator #12: Suggestion to put cameras higher

Sodamin, 2023-06-14:
Ferdinand and me would like to propose to put the cameras of the vehicle higher. When trying to identify traffic lights or anything other than the road, the location of the cameras is suboptimal (esp. if you drive behind e.g. a bus with this Ferrari)

[image]

We would like to suggest to put the cameras similarly to Tesla and other companies on top of the windshield. This is mainly about the middle front facing camera, but probably all 3 can be moved. 

[image]

We have not tested this location yet so cannnot be sure of up or downsides. We should probably position the cameras rather like where they would be if we have a normal, and not a race car.

Einholz-Gao, 2023-06-15:
You are right. We can try to use a normal car rather than a race car later. We have some other Car Models in that Project. Just See here [here](https://www.youtube.com/watch?v=sJR9cDwmbGQ) and you can have a look how can we add a NWH physic engine onto a car model.

Einholz-Gao, 2023-06-15:
after we can drive another normal car but not the default race car.

pranav-ap, 2023-06-29:
Hey, Are the traffic lights in the sim images clear enough to reasonably try and predict the signal? This is the first time I am working on this topic, so I am curious.

[image]

what has been your experience, @Sodamin ?

closed 2023-07-12 by Einholz-Gao