tum-autonomousdriving/autonomous-driving-simulator #19: Reversing does not work when using mouse

pranav-ap, 2023-06-29:
Hey,
I noticed that reversing does not work when using mouse for movement. Is there any reason for designing it this way?
I find it difficult to collect data due to this. Has anyone else had this problem? 

OS: Windows

Thanks

Sodamin, 2023-06-30:
More out of curiosity, when are you going in reverse in your recordings?

pranav-ap, 2023-06-30:
For example, if the car moves onto the sidewalk by accident, it may be useful to record the car moving out of the sidewalk. This could be useful for training?

Sodamin, 2023-06-30:
Hmmm, I don't really know, maybe a question to our supervisor... I'd rather try to keep the model  "clean" and not add that to its training data... (after all the coal would be to keep it off the sidewalk at any time). Esp as the sidewalk can very quickly lead to getting stuck...

pranav-ap, 2023-06-30:
Okay. Got it.

closed 2023-06-30 by pranav-ap