tum-autonomousdriving/autonomous-driving-simulator #21: Traffic Lights Image Clarity

pranav-ap, 2023-06-30:
Hey,

I have not tried to perform traffic light classification yet. I just wanted to know other's experience.

The traffic lights from the dataset don't seem clear enough to make reliable predictions. The signal only seems to become clear once you get quite close to the signal. 

What do you think?

[image]

Thanks

MichaelSodamin, 2023-07-01:
It's right at the edge... we are able to just about early enough identify the light and stop before entering the cross section, but required some work. I was afraid the performance drops more if we render out larger images.

Cars like Teslas apparently use different focal lengths / zoom levels of their forward facing cameras, that would maybe also be a good addition. (but again hard on weaker hardware maybe)

Do you want to give it a go with the current version, it is for sure possible. (and tbh it surprised me as well)

pranav-ap, 2023-07-01:
Okay, I'll try it out!

pranav-ap, 2023-07-01:
I have a general doubt. When we can see multiple traffic signals, how do can we train a model to choose the right one in order to make a decision?

closed 2023-07-12 by pranav-ap