tum-autonomousdriving/autonomous-driving-simulator #23: [Test Simulator] Steering angle not working

Aristotelis1, 2023-07-14:
It seems that the steering value that is passed to the `send_control` is ignored. 
In my case, the car is always going straight.

System: Mac M1

phenschke, 2023-07-16:
Same here on Windows v0.23.

ge49nuk, 2023-07-17:
> Same here on Windows v0.23.

same here

Sodamin, 2023-07-19:
I just downloaded the newest version from gitlab (0.23) windows and it works (steering, breaking and throttle).
Can you show what you are sending exactly and to which IP?

Does any of the 3 controls work?

TuNeXiZz, 2023-07-19:
Well, the `send_control` function works if called in the `vehicle_command` function but not in the `on_image` function,  where I use my model to predict the steering_commands based on the images.

phenschke, 2023-07-19:
Throttle works for me. Steering and brake always stay 0 no matter what value I provide. (e.g, `send_control(0.5, 0.75, 0.1)` in `onimage` makes the car drive with steering angle 0 and throttle 0.75 and brakes 0.)

Sodamin, 2023-07-19:
Can you provide your code what you send to the simulation? 

(The code provided in this repo does not include brake... @Einholz-Gao who is responsible for updating that one? ) 
[image]

Sodamin, 2023-07-20:
I did some digging and it seems when the data is asnycronously sent back to the received images (so more or less data sent then images received) there seems to be a problem (also for me goes into full throttle no steering). Do you observe the same?

Sodamin, 2023-07-20:
Found a workaround, hope it also works for you:
Send your steering controls everytime you get a new image. If your model or PC is not fast enough, take the last controls, but still send them. works for me in the newest version - after I had the same issue as you described

Einholz-Gao, 2023-07-20:
The py code will be updated. We test it and the steering command is working now.

Einholz-Gao, 2023-07-20:
But Attention: here the steering command should be **-1 ~ 1**, if you send 30 or 60, the car will not have any response.

closed 2023-07-29 by Aristotelis1