tum-autonomousdriving/autonomous-driving-simulator #25: Windows Build v0.24

pranav-ap, 2023-07-22:
Hey, 
When will the windows build for v0.24 be uploaded?
Thanks!

Sodamin, 2023-07-22:
Maybe you can help me check out if this version works as expected on Windows: https://drive.google.com/file/d/1Y-rzQxAlZEkeNZB0RbqP9Jrt6lc2vMHJ/view?usp=drive_link

pranav-ap, 2023-07-22:
I noticed the  following issues,

1. The menu are not aligned.
2. Once I go into train mode after pressing the train button, the car controls do not work

[image]

Sodamin, 2023-07-22:
Interesting that the window resized smaller. (can you resize it?)

pranav-ap, 2023-07-22:
No, I cannot

Sodamin, 2023-07-22:
Tried to fix it all: https://drive.google.com/file/d/1XkrZCmOnyZno91qeEYNEEgKzhrm31vjz/view?usp=drive_link

- Weird that it did not drive for you... changed some stuff, works for me, also train mode now
- Also resized the UI elements and you should now be able to resize it later (window starts way bigger for me)

pranav-ap, 2023-07-22:
Yup, Now I'm able to control the vehicle and UI is also proper when I open.

Sodamin, 2023-07-22:
Great!

Small update: (record icon initialized as active, but should have been off)
https://drive.google.com/file/d/1c8ZGFmXQltKOaCm14XrKahtA6FC9AgUZ/view?usp=drive_link

closed 2023-08-10 by pranav-ap