tum-autonomousdriving/autonomous-driving-simulator #26: IP Doubt

pranav-ap, 2023-07-22:
Hey,

What is the below line supposed to be while using the v0.23 simulator in windows?

`eventlet.wsgi.server(eventlet.listen(("", 4567)), app) `

The IP text file says : 192.168.2.224

The car is not moving.

Thanks

Sodamin, 2023-07-22:
The line in the code should be fine (we have the same)
the IP in the file should be changed to your IP

Sodamin, 2023-07-22:
Actually you can just put in "localhost" (without " )

pranav-ap, 2023-07-22:
I get the following line, and thats it. No change after that.

[image]

Kinda weird.

Sodamin, 2023-07-22:
so far so good. I have the same, when I then start the simulator the python starts to receive images.

What might be a problem are your firewall / network settings. I am not sure what to do exactly but you must make sure the program is allowed to be reached... I think there were similar problems on the other simulator (from udacity) maybe you find material on that

closed 2023-08-10 by pranav-ap